atc1441/NETSGPClient #3: Issu with SG500md

anth039, 2021-07-30:
Hello,

I did the test for my SG500MD, it seems that I have strange values (see pj).

400w of panel with two panels on each side of 16v (that is to say 32v by entry)

[https://ibb.co/x5M0z1X](url)

ESP32 with LC12S

16-17-4 => RX - TX - SET of LC12S
CS => GROUND of LC12S

Sorry for my speak english

enwi, 2022-01-08:
Hey could you try to get a dump of the received data?

For that add the following after the shown line:
https://github.com/atc1441/NETSGPClient/blob/42189065c091018a775a8bf41f9c713e9931040a/NETSGPClient.cpp#L20
```
Serial.printf("%032X", mBuffer);
```

anth039, 2022-01-15:
Hello
@enwi
[image]

enwi, 2022-01-19:
@anth039 Okay I think I messed up the print statement. Could you try again using the follwing code instead of `Serial.printf("%032X", mBuffer);`?

```c++
for (uint8_t i = 0; i < 32; i++)
{
    Serial.printf("%02X", mBuffer[i]);
}
Serial.printf("\n");
```

Not to myself: Add more debug code in general

anth039, 2022-01-19:
@enwi Hello, 

with instead of Serial.printf("%032X", mBuffer);
[image]

enwi, 2022-01-24:
@anth039 Could you also provide your InverterID? Or is it `34000284`?

Do you also have more test data for me than `43C00000000034000284800000000A712008F0000000000000000000000000000`?

anth039, 2022-01-28:
Hello @enwi , 34002848 is my inverterID.

"
constexpr const uint8_t PROG_PIN = 4; /// Programming enable pin of RF module
constexpr const uint8_t RX_PIN = 16; /// RX pin of RF module
constexpr const uint8_t TX_PIN = 17; /// TX pin of RF module
constexpr const uint32_t inverterID = 0x34002848; /// Identifier of your inverter (see label on inverter)
"
I have keep your code, i have only change inverter ID.

what i can to make for to get more data 
[image]
?

enwi, 2022-01-28:
@anth039 thanks for the update. It seems that maybe for the `MD` inverters the data layout is somewhat different than the `MS` ones.

For the data just let the program run a little longer and then post a list of the debugged status like `43C00000000034000284800000000A712008F0000000000000000000000000000`. But please copy and paste it here, so I don't have to retype it 😅.

@atc1441 Seems like this inverter is sending data differently?

atc1441, 2022-01-28:
@enwi As you mention best would be to habe more data to reverse. At best with the real values as well to have something to compare.

The the easiest is to put the SG500md and SG200md data next to each other and compare them.

anth039, 2022-02-05:
Hello,

@enwi 

"
Sending request now
43C0000000003400284800000000A712008F0000000000000000000000000000
*********************************************
Received Inverter Status
Device: 34002848
Status: 0
DC_Voltage: 46.00V
DC_Current: 366.00A
DC_Power: 16836.00W
AC_Voltage: 0.00V
AC_Current: 0.00A
AC_Power: 0.00W
Power gen total: 0.00
Temperature: 0

Sending request now
43C0000000003400284800000000A712008F0000000000000000000000000000
*********************************************
Received Inverter Status
Device: 34002848
Status: 0
DC_Voltage: 46.00V
DC_Current: 366.00A
DC_Power: 16836.00W
AC_Voltage: 0.00V
AC_Current: 0.00A
AC_Power: 0.00W
Power gen total: 0.00
Temperature: 0

Sending request now

Sending request now

Sending request now
43C0000000003400284800000000A712008F0000000000000000000000000000
*********************************************
Received Inverter Status
Device: 34002848
Status: 0
DC_Voltage: 46.00V
DC_Current: 366.00A
DC_Power: 16836.00W
AC_Voltage: 0.00V
AC_Current: 0.00A
AC_Power: 0.00W
Power gen total: 0.00
Temperature: 0

Sending request now
43C0000000003400284800000000A712008F0000000000000000000000000000
*********************************************
Received Inverter Status
Device: 34002848
Status: 0
DC_Voltage: 46.00V
DC_Current: 366.00A
DC_Power: 16836.00W
AC_Voltage: 0.00V
AC_Current: 0.00A
AC_Power: 0.00W
Power gen total: 0.00
Temperature: 0

Sending request now
[... 192 lines cut ...]
DC_Current: 366.00A
DC_Power: 16836.00W
AC_Voltage: 0.00V
AC_Current: 0.00A
AC_Power: 0.00W
Power gen total: 0.00
Temperature: 0

Sending request now
43C0000000003400284800000000A712008F0000000000000000000000000000
*********************************************
Received Inverter Status
Device: 34002848
Status: 0
DC_Voltage: 46.00V
DC_Current: 366.00A
DC_Power: 16836.00W
AC_Voltage: 0.00V
AC_Current: 0.00A
AC_Power: 0.00W
Power gen total: 0.00
Temperature: 0
"

Regards

enwi, 2022-02-06:
@anth039 Thanks for the input, but it seems that the received status does not change much. Do you know at what time of day you made the logs? It would be most helpful to have data where the solar input power and temperature as well as other parameters sent by the inverter change.

anth039, 2022-02-15:
@enwi Hello,

The log still look the same.

ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:10944
load:0x40080400,len:6388
entry 0x400806b4
Welcome to Micro Inverter Interface by ATCnetz.de and enwi
Could not set RF module to default settings

Sending request now
43C0000000003400284800000000A712008F0000000000000000000000000000
*********************************************
Received Inverter Status
Device: 34002848
Status: 0
DC_Voltage: 46.00V
DC_Current: 366.00A
DC_Power: 16836.00W
AC_Voltage: 0.00V
AC_Current: 0.00A
AC_Power: 0.00W
Power gen total: 0.00
Temperature: 0

Sending request now
43C0000000003400284800000000A712008F0000000000000000000000000000
*********************************************
Received Inverter Status
Device: 34002848
Status: 0
DC_Voltage: 46.00V
DC_Current: 366.00A
DC_Power: 16836.00W
AC_Voltage: 0.00V
AC_Current: 0.00A
AC_Power: 0.00W
Power gen total: 0.00
Temperature: 0

Sending request now
43C0000000003400284800000000A712008F0000000000000000000000000000
*********************************************
Received Inverter Status
Device: 34002848
Status: 0
DC_Voltage: 46.00V
DC_Current: 366.00A
DC_Power: 16836.00W
AC_Voltage: 0.00V
AC_Current: 0.00A
AC_Power: 0.00W
Power gen total: 0.00
Temperature: 0

Sending request now
43C0000000003400284800000000A712008F0000000000000000000000000000
*********************************************
Received Inverter Status
Device: 34002848
Status: 0
[... 8648 lines cut ...]
AC_Current: 0.00A
AC_Power: 0.00W
Power gen total: 0.00
Temperature: 0

Sending request now

Sending request now

Sending request now

Sending request now
43C0000000003400284800000000A712008F0000000000000000000000000000
*********************************************
Received Inverter Status
Device: 34002848
Status: 0
DC_Voltage: 46.00V
DC_Current: 366.00A
DC_Power: 16836.00W
AC_Voltage: 0.00V
AC_Current: 0.00A
AC_Power: 0.00W
Power gen total: 0.00
Temperature: 0

enwi, 2022-02-20:
@anth039 I have shortened your comment since the same data is repeated "endlessly". Though I think it will be hard to extract any information here, since the received data does not change at all. 

@atc1441 Do you have an idea what could be done?

atc1441, 2022-02-20:
Maybe a solution to get more differend data would be too request it at differend times over the day. Like with a lot of sun, bit less sun etc.

If the data changes at all ?!

Some real data from the same moments are more or less mendatory. For example an voltmeter / ampmeter in the solar line. (Be carefull with any high voltage!)

enwi, 2022-02-26:
@anth039 That does seem a lot better now. Also the data seems resonable.

I have to add that `AC current`, `AC power` and `Power gen total` don`t seem to be correct

enwi, 2022-02-26:
I think it would also make sense to check the CRC and only use messages that are fine

anth039, 2022-02-27:
Hello @enwi 

thank you for your work

There seems to be a problem with the CRC, no?

Only the temperature seems incorrect to me now ^^

Sending request now
43C0000000003400284842934B00C70AFF00B358D30014000040950000000000
CRC is *********************************************
Received Inverter Status
Device: 34002848
Status: 64
DC_Voltage: 28.15V
DC_Current: 1.79A
DC_Power: 50.39W
AC_Voltage: 227.39V
AC_Current: 0.20A
AC_Power: 45.48W
Power gen total: 73.65
Temperature: 149

Sending request now
43C0000000003400284842934B00C70AFF00B358FD0014000040950000000000
CRC is *********************************************
Received Inverter Status
Device: 34002848
Status: 64
DC_Voltage: 28.15V
DC_Current: 1.79A
DC_Power: 50.39W
AC_Voltage: 227.81V
AC_Current: 0.20A
AC_Power: 45.56W
Power gen total: 73.65
Temperature: 149

Sending request now
43C0000000003400284842934B00C70AFF00B358FD0014000040950000000000
CRC is *********************************************
Received Inverter Status
Device: 34002848
Status: 64
DC_Voltage: 28.15V
DC_Current: 1.79A
DC_Power: 50.39W
AC_Voltage: 227.81V
AC_Current: 0.20A
AC_Power: 45.56W
Power gen total: 73.65
Temperature: 149

Sending request now
43C0000000003400284842934B00C70AFF00B358D30014000040950000000000
CRC is *********************************************
Received Inverter Status
Device: 34002848
Status: 64
DC_Voltage: 28.15V
DC_Current: 1.79A
DC_Power: 50.39W
AC_Voltage: 227.39V
AC_Current: 0.20A
AC_Power: 45.48W
Power gen total: 73.65
Temperature: 149

Sending request now
43C0000000003400284842934B00C70AFF00B358D30014000040950000000000
CRC is *********************************************
Received Inverter Status
Device: 34002848
Status: 64
DC_Voltage: 28.15V
DC_Current: 1.79A
DC_Power: 50.39W
AC_Voltage: 227.39V
AC_Current: 0.20A
AC_Power: 45.48W
Power gen total: 73.65
Temperature: 149

Sending request now
43C0000000003400284842934B00C70AFF00B658FD0014000040950000000000
CRC is *********************************************
Received Inverter Status
Device: 34002848
Status: 64
DC_Voltage: 28.15V
DC_Current: 1.82A
DC_Power: 51.23W
AC_Voltage: 227.81V
AC_Current: 0.20A
AC_Power: 45.56W
Power gen total: 73.65
Temperature: 149

Sending request now
43C0000000003400284842934B00C70AF400B358FD0014000040950000000000
CRC is *********************************************
Received Inverter Status
Device: 34002848
Status: 64
DC_Voltage: 28.04V
DC_Current: 1.79A
DC_Power: 50.19W
AC_Voltage: 227.81V
AC_Current: 0.20A
AC_Power: 45.56W
Power gen total: 73.65
Temperature: 149

Sending request now
43C0000000003400284842934B00C70AFF00B358D30014000040950000000000
CRC is *********************************************
Received Inverter Status
Device: 34002848
Status: 64
DC_Voltage: 28.15V
DC_Current: 1.79A
DC_Power: 50.39W
AC_Voltage: 227.39V
AC_Current: 0.20A
AC_Power: 45.48W
Power gen total: 73.65
Temperature: 149

Sending request now
43C0000000003400284842934B00C70AFF00B358FD0014000040950000000000
CRC is *********************************************
Received Inverter Status
Device: 34002848
Status: 64
DC_Voltage: 28.15V
DC_Current: 1.79A
DC_Power: 50.39W
AC_Voltage: 227.81V
AC_Current: 0.20A
AC_Power: 45.56W
Power gen total: 73.65
Temperature: 149

Sending request now
43C0000000003400284842934B00C70AFF00B358FD0014000040950000000000
CRC is *********************************************
Received Inverter Status
Device: 34002848
Status: 64
DC_Voltage: 28.15V
DC_Current: 1.79A
DC_Power: 50.39W
AC_Voltage: 227.81V
AC_Current: 0.20A
AC_Power: 45.56W
Power gen total: 73.65
Temperature: 149

Sending request now
43C0000000003400284842934B00C70AFF00B358FD0014000040950000000000
CRC is *********************************************
Received Inverter Status
Device: 34002848
Status: 64
DC_Voltage: 28.15V
DC_Current: 1.79A
DC_Power: 50.39W
AC_Voltage: 227.81V
AC_Current: 0.20A
AC_Power: 45.56W
Power gen total: 73.65
Temperature: 149

Sending request now
43C0000000003400284842934B00C70AFF00B359260014000040950000000000
CRC is *********************************************
Received Inverter Status
Device: 34002848
Status: 64
DC_Voltage: 28.15V
DC_Current: 1.79A
DC_Power: 50.39W
AC_Voltage: 228.22V
AC_Current: 0.20A
AC_Power: 45.64W
Power gen total: 73.65
Temperature: 149

Sending request now
43C0000000003400284842934B00C70AF400B358D30014000040950000000000
CRC is *********************************************
Received Inverter Status
Device: 34002848
Status: 64
DC_Voltage: 28.04V
DC_Current: 1.79A
DC_Power: 50.19W
AC_Voltage: 227.39V
AC_Current: 0.20A
AC_Power: 45.48W
Power gen total: 73.65
Temperature: 149

Sending request now
43C0000000003400284842934B00C70AFF00B358FD0014000040950000000000
CRC is *********************************************
Received Inverter Status
Device: 34002848
Status: 64
DC_Voltage: 28.15V
DC_Current: 1.79A
DC_Power: 50.39W
AC_Voltage: 227.81V
AC_Current: 0.20A
AC_Power: 45.56W
Power gen total: 73.65
Temperature: 149

anth039, 2022-02-27:
Hello,

its good for me, and you ?

sending request now
43C000000000340028484293D700530AC900435B46000600003F060000000000
CRC is valid
*********************************************
Received Inverter Status
Device: 34002848
Status: 63
DC_Voltage: 27.61V
DC_Current: 0.67A
DC_Power: 18.50W
AC_Voltage: 233.66V
AC_Current: 0.06A
AC_Power: 14.02W
Power gen total: 73.92
Temperature: 6

Sending request now
43C000000000340028484293D700530AC900435B46000600003F070000000000
CRC is valid
*********************************************
Received Inverter Status
Device: 34002848
Status: 63
DC_Voltage: 27.61V
DC_Current: 0.67A
DC_Power: 18.50W
AC_Voltage: 233.66V
AC_Current: 0.06A
AC_Power: 14.02W
Power gen total: 73.92
Temperature: 7

Sending request now
43C000000000340028484293D700530AC900435B46000600003F060000000000
CRC is valid
*********************************************
Received Inverter Status
Device: 34002848
Status: 63
DC_Voltage: 27.61V
DC_Current: 0.67A
DC_Power: 18.50W
AC_Voltage: 233.66V
AC_Current: 0.06A
AC_Power: 14.02W
Power gen total: 73.92
Temperature: 6

Sending request now
43C000000000340028484293D700530AC900435B1C000600003F070000000000
CRC is valid
*********************************************
Received Inverter Status
Device: 34002848
Status: 63
DC_Voltage: 27.61V
DC_Current: 0.67A
DC_Power: 18.50W
AC_Voltage: 233.24V
AC_Current: 0.06A
AC_Power: 13.99W
Power gen total: 73.92
Temperature: 7

Sending request now

Sending request now
43C000000000340028484293D700530AC900435B46000600003F070000000000
CRC is valid
*********************************************
Received Inverter Status
Device: 34002848
Status: 63
DC_Voltage: 27.61V
DC_Current: 0.67A
DC_Power: 18.50W
AC_Voltage: 233.66V
AC_Current: 0.06A
AC_Power: 14.02W
Power gen total: 73.92
Temperature: 7

Sending request now
43C000000000340028484293D700530AC900435B46000600003F060000000000
CRC is valid
*********************************************
Received Inverter Status
Device: 34002848
Status: 63
DC_Voltage: 27.61V
DC_Current: 0.67A
DC_Power: 18.50W
AC_Voltage: 233.66V
AC_Current: 0.06A
AC_Power: 14.02W
Power gen total: 73.92
Temperature: 6

Sending request now
43C000000000340028484293D700530AC900405B46000600003F070000000000
CRC is valid
*********************************************
Received Inverter Status
Device: 34002848
Status: 63
DC_Voltage: 27.61V
DC_Current: 0.64A
DC_Power: 17.67W
AC_Voltage: 233.66V
AC_Current: 0.06A
AC_Power: 14.02W
Power gen total: 73.92
Temperature: 7

Sending request now

Sending request now
43C000000000340028484293D700530AC900435B46000600003F070000000000
CRC is valid
*********************************************
Received Inverter Status
Device: 34002848
Status: 63
DC_Voltage: 27.61V
DC_Current: 0.67A
DC_Power: 18.50W
AC_Voltage: 233.66V
AC_Current: 0.06A
AC_Power: 14.02W
Power gen total: 73.92
Temperature: 7

enwi, 2022-02-28:
> its good for me, and you ?

Yeah, then I will close this issue now :)

closed 2022-02-28 by enwi

anth039, 2022-02-28:
Thank you for your work it's perfect