atc1441/NETSGPClient #4: Could not set RF module to default settings

jjtoard, 2021-09-11:
Hi,

I'm very interested in you project and was trying the pol sample on an esp8266
I added some code to let it work

**#include <SoftwareSerial.h>
SoftwareSerial SerialJJ(7, 8);**
NETSGPClient client(SerialJJ, PROG_PIN); /// NETSGPClient instance

and in the setup 
//SerialJJ.begin(9600, SERIAL_8N1, RX_PIN, TX_PIN);
    **SerialJJ.begin(9600);**

I get the following error 

Welcome to Micro Inverter Interface by ATCnetz.de and enwi
Could not set RF module to default settings

Sending request now

Could you help me with this ?

123gerald1, 2021-10-02:
Hello jjtoard,
I would like to realize the query of the inverters also with a Wemos D1 mini.
I would be interested if you have done it with it?

Sg. Gerald

Eule01x, 2022-01-03:
> Hi,
> 
> I'm very interested in your project and was trying the pol sample on an esp8266 wemos d1 mini I added some code to let it work
> 
> **#include <SoftwareSerial.h> SoftwareSerial SerialJJ(D7, D8);** NETSGPClient client(SerialJJ, PROG_PIN); /// NETSGPClient instance
> 
> and in the setup //SerialJJ.begin(9600, SERIAL_8N1, RX_PIN, TX_PIN); **SerialJJ.begin(9600);**
> 
> I get the following error
> 
> ```
> Welcome to Micro Inverter Interface by ATCnetz.de and enwi
> Could not set RF module to default settings
> 
> Sending request now
> ```
> 
> Could you help me with this ?
> 
> I also tried your exact code on an esp32 wroom board. I get the same error. On the LC12S do you connect gnd and cs both to the gnd on the esp ?

I have the same issue.
Using a Lolin32 Board I get the same message. As far as I found out: CS should be tied to GND, the SET pin connects to the "PROG_PIN" from PollDemo.
I tried some other values for module ID (found 0x2233 on the web) or netword ID (found 0x1112 on the web). But without any effect.

Anyway, it seems I can read out something from a Hoymiles converter, but the values are weird and it mostly only works when I fiddle around the cables / the LC12S module. Whether I get a better RF connection or my wires have some loose contacts...

regards, Eule01x

enwi, 2022-01-08:
Hey, sorry for the late reply. Somehow GitHub did not notify me on opened issues for this repo.

I am using the library as well with an ESP8266 and my connections are as follows:
- ESP 3V3 <-> VCC
- ESP TX <-> RX
- ESP RX <-> TX
- ESP D3 <-> SET
- ESP GND <-> CS
- ESP GND <-> GND

closed 2022-02-26 by enwi