atc1441/NETSGPClient #5: Error message during compilation

123gerald1, 2021-10-01:
Hello,
I have tried to compile the program, unfortunately I always get an error message.

Serial2' was not declared in this scope

What could be the reason for this?
Please for your help

atc1441, 2021-10-01:
Hey. 

Sounds like you do not use ESP32 for it. 

It needs a device with 2 serial ports or a software serial instance

123gerald1, 2021-10-01:
Hello,
I would like to realize it either with a WEMOS D1 mini or with a Node MCU.
Is it possible with this?

enwi, 2022-01-08:
Hey @123gerald1 you can use the library with the ESP8266 aka WEMOS D1 mini. So am I.

In the example code you just need to replace `Serial2` with `Serial1` and connect the RF module as follows:
- ESP 3V3 <-> VCC
- ESP TX <-> RX
- ESP RX <-> TX
- ESP D3 <-> SET
- ESP GND <-> CS
- ESP GND <-> GND

https://github.com/atc1441/NETSGPClient/blob/42189065c091018a775a8bf41f9c713e9931040a/examples/PollDemo/PollDemo.ino#L8
https://github.com/atc1441/NETSGPClient/blob/42189065c091018a775a8bf41f9c713e9931040a/examples/AsyncDemo/AsyncDemo.ino#L8

closed 2022-02-26 by enwi