atc1441/NETSGPClient #9: Can not configure LC12S

cody82, 2022-03-11:
I am using an ESP32 and an SG600MD.
I connected the LC12S like that:
    ESP 3V3 <-> VCC
    ESP TX <-> RX
    ESP RX <-> TX
    ESP GPIO4 <-> SET
    ESP GND <-> CS
    ESP GND <-> GND

But I get the message "Could not set RF module to default settings".
Some debugging reveals that after sending the settings to the LC12S, the read command returns only one zero byte instead of expected 18 bytes.

I verified that the RX and TX pins are setup correctly. The LED on the LC12S lights up on reset for ~2 seconds, then it flashes shortly once per second.
What could be wrong here?

enwi, 2022-03-13:
@cody82 What is the code that you used? Do you have a minimal working example?

cody82, 2022-03-13:
I used this code: https://github.com/atc1441/NETSGPClient/blob/main/examples/PollDemo/PollDemo.ino
I only added a few debug messages.

enwi, 2022-03-14:
The connections and everything seem to be fine and I can't reproduce this issue at all. So I don't know if I can help you with this one 😅

closed 2022-07-18 by enwi