atc1441/NETSGPClient #10: Dependencies "StandardCplusplus" not existing ?!

atc1441, 2022-03-15:
While installing the Library from the Library manager it asks to also install also the dependencies which list is empty. 
But would normaly contain the StandardCplusplus which is not in my lists of librarys

[image]

Trying to install it ends up in an error, the "NETSGPClient" Library gets still installed.

[image]

The issue seems that the "StandardCplusplus" lib is not in the Arduino library list so will fail to install

Should we maybe remove it ?

enwi, 2022-03-15:
I think we should remove it, because `StandardCplusplus` is not an "official" arduino library (bc it does not have the `library.properties` file), which causes this exception, because the library manager cannot find it.

snelweg, 2022-07-16:
While installing the NETSGPClient I still get the missing StandardCplusplus error.

enwi, 2022-07-18:
That is to be expected, because I haven't removed it from the `library.properties` file yet, hence the issue still being open

commit d0300c79 (2022-07-18): Remove StandardCplusplus from properties closes #10
commit 8a8e41e8 (2022-07-19): Remove StandardCplusplus from properties closes #10
closed 2022-07-19 by enwi