atc1441/NETSGPClient #11: question: is it possible to use an nRF24L01

heiko1988, 2022-05-26:
Good day,

sorry im a noob, in this tech ;)

Can use an nRF24L01 in combination with an esp32 to query the data?

If it is possible, how do I have to adapt the script and connect the nRF24L01 to the esp32?

Thanks very much :)

enwi, 2022-05-31:
Why would you want to use an nRF24L01?
Is this related to #6?

heiko1988, 2022-05-31:
The problem is, the lc12s delivery time 1 months, and the nRF24L01 I have home.

When there is no way. Do I have to wait until the lc12 is there ;)

enwi, 2022-05-31:
Theoretically it should be possible to make it work, thought the software might need to be refactore. I am thinking of the `Strategy`-pattern

heiko1988, 2022-05-31:
OK. Is it difficult to customize? I'm not that good at it. But would like to help

NRF24L01 has spi protocol 
Here a lot of information 
https://www.mikrocontroller.net/articles/NRF24L01_Tutorial

enwi, 2022-05-31:
There is some work to be done, but I currently don't have that much time to do it. I have actually worked with the `NRF24L01+` a lot before and I would suggest to use [this library](https://github.com/nRF24/RF24) to control it.

enwi, 2022-05-31:
> Are they compatible wireless protocol wise?
> Does the nRF24L01 even have a "Raw" radio RX mode? I reccon it was only compatible to Nordics proprietary one but that could be tottaly wrong.

That's why I asked 🤣

heiko1988, 2022-05-31:
That's a good question, not familiar with the specs. Just have no experience with it. It will probably be best to wait for the ls12. Thank you very much for your information.

heiko1988, 2022-05-31:
Yes I have buy for 2 weeks, delivery time 4-6 weeks...

atc1441, 2022-05-31:
Where are you from?

heiko1988, 2022-05-31:
From Germany

atc1441, 2022-05-31:
Nice can send you some. Please send a mail to heiko@43u.de

heiko1988, 2022-05-31:
Perfect, have send you a mail

closed 2022-05-31 by enwi