atc1441/NETSGPClient #13: Try too use with mqtt

heiko1988, 2022-06-04:
Good day,

I'm trying to integrate a mqtt client that I can use the data in the smart home. in my case it would be iobroker

I've tried a bit to integrate it. Unfortunately, I lack expertise

I hope one of you can help me

The demo with the serial output runs great :)

My hardware:

Esp32 Wroom

The code where not working:

#include "NETSGPClient.h"
#include "EspMQTTClient.h"

EspMQTTClient client(
  "test",
  "test",
  "192.168.0.119",  // MQTT Broker server ip
  "iobroker",   // Can be omitted if not needed
  "Passwort",   // Can be omitted if not needed
  "SG600MD"      // Client name that uniquely identify your device
);

constexpr const uint8_t PROG_PIN = 4; /// Programming enable pin of RF module
constexpr const uint8_t RX_PIN = 17; /// RX pin of RF module
constexpr const uint8_t TX_PIN = 16; /// TX pin of RF module
constexpr const uint32_t inverterID = 0x38004044; /// Identifier of your inverter (see label on inverter)

NETSGPClient client(Serial2, PROG_PIN); /// NETSGPClient instance

void setup()
{
    Serial.begin(115200);
    Serial2.begin(9600, SERIAL_8N1, RX_PIN, TX_PIN);
//    pinMode(LED_BUILTIN, OUTPUT);
    delay(1000);
    Serial.println("Welcome to Micro Inverter Interface by ATCnetz.de and enwi.one");

    // Make sure the RF module is set to the correct settings
    if (!client.setDefaultRFSettings())
    {
        Serial.println("Could not set RF module to default settings");
    }
}

void onConnectionEstablished() {

  client.subscribe("mytopic/test", [] (const String &payload)  {
    Serial.println(payload);
  });

  client.publish("mytopic/test", "This is a message");
}

uint32_t lastSendMillis = 0;
void loop()
{
    const uint32_t currentMillis = millis();
    if (currentMillis - lastSendMillis > 2000)
    {
        lastSendMillis = currentMillis;

//        digitalWrite(LED_BUILTIN, !digitalRead(LED_BUILTIN));
        Serial.println("");
        Serial.println("Sending request now");

        const NETSGPClient::InverterStatus status = client.getStatus(inverterID);
        if (status.valid)
        {
            Serial.println("*********************************************");
            Serial.println("Received Inverter Status");
            Serial.print("Device: ");
            Serial.println(status.deviceID, HEX);
            Serial.println("Status: " + String(status.state));
            Serial.println("DC_Voltage: " + String(status.dcVoltage) + "V");
            Serial.println("DC_Current: " + String(status.dcCurrent) + "A");
            Serial.println("DC_Power: " + String(status.dcPower) + "W");
            Serial.println("AC_Voltage: " + String(status.acVoltage) + "V");
            Serial.println("AC_Current: " + String(status.acCurrent) + "A");
            Serial.println("AC_Power: " + String(status.acPower) + "W");
            Serial.println("Power gen total: " + String(status.totalGeneratedPower));
            Serial.println("Temperature: " + String(status.temperature));
        }
    }
}

the error description from the compiler

PollDemo:21:21: error: conflicting declaration 'NETSGPClient client'
 NETSGPClient client(Serial2, PROG_PIN); /// NETSGPClient instance
                     ^
C:\Users\chaos\Desktop\NETSGPClient-1.1.0 (1)\NETSGPClient-1.1.0\examples\PollDemo\PollDemo.ino:4:15: note: previous declaration as 'EspMQTTClient client'
 EspMQTTClient client(
               ^
C:\Users\chaos\Desktop\NETSGPClient-1.1.0 (1)\NETSGPClient-1.1.0\examples\PollDemo\PollDemo.ino: In function 'void setup()':
PollDemo:32:17: error: 'class EspMQTTClient' has no member named 'setDefaultRFSettings'
     if (!client.setDefaultRFSettings())
                 ^
C:\Users\chaos\Desktop\NETSGPClient-1.1.0 (1)\NETSGPClient-1.1.0\examples\PollDemo\PollDemo.ino: In function 'void loop()':
PollDemo:62:60: error: 'class EspMQTTClient' has no member named 'getStatus'
         const NETSGPClient::InverterStatus status = client.getStatus(inverterID);
                                                            ^
Mehrere Bibliotheken wurden für "WiFi.h" gefunden
 Benutzt: C:\Users\chaos\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6\libraries\WiFi
 Nicht benutzt: C:\Program Files (x86)\Arduino\libraries\WiFi
exit status 1
conflicting declaration 'NETSGPClient client'

enwi, 2022-06-06:
@heiko1988 you cannot have two variables/objects with the same name. Rename the `EspMQTTClient` variable `client` in line 4 to maybe `mqtt` and also everywhere else in the code where you used it, but don't rename the `client` variables associated with `NETSGPClient`

heiko1988, 2022-06-06:
Thank you for your important information

I put something together from different tuts.
The result is a small system that meets my requirements

With the following Things

Small web interface
http://your_ip

[image]

web Serial
http://your_ip/webserial

[image]

Mqtt

[image]

The coed isn't the best and just thrown together, but it's enough for me :)

Code:
#include "NETSGPClient.h"
#include <Arduino.h>
#include <WiFi.h>
#include <AsyncTCP.h>
#include <ESPAsyncWebServer.h>
#include <WebSerial.h>
#include <PubSubClient.h>
#define LED 2

//Optionen Start
//Add your Wifi Data
const char* ssid = "Wlan@Home_";           // Your WiFi SSID
const char* password = "11Reich22!";      // Your WiFi Password

// Add your MQTT Broker Data:
const char* mqtt_server = "192.168.0.119";  //mqtt Broker ip
const char* mqtt_user = "iobroker";         //mqtt Username
const char* mqtt_password = "iobroker";     //mqtt Password
//Optionen End

// Create AsyncWebServer object on port 80
AsyncWebServer server(80);

// Create an Event Source on /events
AsyncEventSource events("/events");

// Timer variables
unsigned long lastTime = 0;  
unsigned long timerDelay = 30000;

WiFiClient espClient;
PubSubClient mqtt(espClient);
long lastMsg = 0;
char msg[50];
int value = 0;

//Variable für mqtt
float temperature = 0;
float dcVoltage1 = 0;
float dcCurrent1 = 0;
float dcPower1 = 0;
float acVoltage1 = 0;
float acCurrent1 = 0;
float acPower1 = 0;
float temperature1 = 0;
float totalGeneratedPower1 = 0;
[... 402 lines cut ...]
    //Serial.println(tempString);
    mqtt.publish("pv1/temperature", tempString);

    totalGeneratedPower1 = (status.totalGeneratedPower);
    char totalGeneratedPower[8];
    dtostrf(totalGeneratedPower1, 1, 2, totalGeneratedPower);
    //Serial.print("Mqtt-totalGeneratedPower: ");
    //Serial.println(totalGeneratedPower);
    mqtt.publish("pv1/totalGeneratedPower", totalGeneratedPower);

    state1 = (status.state);
    char state[8];
    dtostrf(state1, 1, 2, state);
    //Serial.print("Mqtt-state: ");
    //Serial.println(state);
    mqtt.publish("pv1/state", state);
    }   

  }
}

Quellen:
https://randomnerdtutorials.com/esp32-webserial-library/
https://randomnerdtutorials.com/esp32-web-server-sent-events-sse/
https://randomnerdtutorials.com/esp32-mqtt-publish-subscribe-arduino-ide/

enwi, 2022-06-06:
@heiko1988 I hope the WiFi credentials are just an example 😅 

Other than that looks great. I am actually working on a PCB and webinterface for these inverters, which I will sell on Tindie soon. It will have the ability to provide data to MQTT, PVOutput and to configure 1 or more inverters through a webinterface.

closed 2022-06-06 by enwi

WeareGermans, 2022-08-03:
Maybe this could be transformed into a great example?

enwi, 2022-08-03:
Pull requests welcome @WeareGermans 😜

MichaelHeimann, 2023-01-18:
> Maybe this could be transformed into a great example?

I got so inspired by this gem that I (although not having coding-centric background) created my first github project based on this "example". But I did use a ESP32 (future ready for bluetooth, maybe) as a base:

https://github.com/MichaelHeimann/OpenNETek

I have changed it quite a bit but would like to thank you guys for the inspiration!

enwi, 2023-01-19:
Thanks for the nice work @MichaelHeimann. Though I got one improvement for your repo. Rather than copying the NETSGPClient code into your repo you should add it as a dependency.

MichaelHeimann, 2023-01-19:
Thanks @enwi for pointing that out, it is changed.