atc1441/NETSGPClient #14: Solder an ESP directly to the inverter

majuss, 2022-06-11:
As far as I understood it, the wireless modules creating a raw bridge. So it would be possible to remove the rf module from the inverter and directly replace it with an ESP32? The only requirement would be, that the 3.3 V LDO is delivering enough current.

majuss, 2022-06-11:
[image]
So under the RF module is the LDO. A AMS1117 which can handle 1 A and is also used on the ESP32 WROOM board. So I think this shouldn't be a problem at all! I measured across the whole inverter board. The 3.3 V of this LDO is not used anywhere else.

majuss, 2022-06-11:
So the ESP now is connected to the Inverter and I successfully uploaded the code. I'am not getting any valid status' :( Is the wireless module adding a layer of abstraction which I first of to remove from the code?

enwi, 2022-06-11:
@majuss yes the RF module is adding a layer of abstraction to send the data over the air, but that is just the modulation of the transmitted signal. So once  another module receives the modulated data it demodulates it, so the module will spit out the same bytes you put in to the sender.

enwi, 2022-06-11:
Also what makes you think that?
> The only requirement would be, that the 3.3 V LDO is delivering enough current.

majuss, 2022-06-11:
Yeah you're right, maybe I need a level shifter, when the inverter is communicating with 5 V on tx and rx. But the rf module only survives 3.6 V, so it would be kinda surprising if it's running the serial interface on 5 V.

Could you give me hint what code I need to delete? I will dig into it tomorrow.

enwi, 2022-06-12:
I don't think you need to modify the source code, in the examples you just need to remove or comment the following lines:
```c++
// Make sure the RF module is set to the correct settings
    if (!client.setDefaultRFSettings())
    {
        Serial.println("Could not set RF module to default settings");
    }
```

majuss, 2022-06-12:
Thank you! I commented that out, but unfortunatley I still only get "Sending request now" and nothing else. I connected 26 V DC to the Inverter.

majuss, 2022-06-12:
So I found out, that the RX and TX of the RF module is going to an PIC16F1936 which operates at 5 V with a 78M05 LDO. That suggests that the serial is indeed 5 V, but there is a 4.7 k resistor in series on the TX line. Which means the RF module is not recieving 5 but 3.3 V (at 0.35 mA). This resistor is missing on the RX line, which only has a pulldown of 10 k. So I won't need a level shifter in my opinion, since the RF module only runs at 3.3 V operating voltage, it has no way of sending 5 V signals.

enwi, 2022-06-12:
A series resistor can only limit the current, the voltage will only be affected slightly. A voltage divider can drop the voltage, but is also not ideal. Though I haven't looked into the schematic of the inverter and the code of the PIC is also closed source.

But I would say you would be fine connecting an ESP8266 directly as it is 5V tolerant, but I am not sure about an ESP32.

majuss, 2022-06-12:
I think both are not 5 V tolerant :D

So I desoldered the RF module of one of my SG800MD's and connected it to my ESP32. Entered the inverter id and prog pin. The ESP is able to set the default values for the rf module and it is blinking every 2 seconds. But I'am still getting no data :/

[image]

enwi, 2022-06-12:
> I think both are not 5 V tolerant :D
https://hackaday.com/2016/07/28/ask-hackaday-is-the-esp8266-5v-tolerant/

enwi, 2022-06-12:
@majuss is the RF module blinking twice or only once every 2 seconds?

majuss, 2022-06-12:
It only blinks once.

enwi, 2022-06-12:
Okay that means that the inverter is not replying to the request. Then I would suggest to double check the inverter ID and if it is correctly set in the code

enwi, 2022-06-14:
You can theoretically brute force it by making a loop through all possible identifiers and sending a request, if you get a reply you know the ID. Luckily you can reduce the number of IDs to serach, because I think each model starts with a different constant ID. Though I don't know the starting ID for you model `SG800MD`, because I could not find a lot of information online. Your ID was `0x45002819` right? So you should search all IDs starting from `0x45000001` up to `0x4500FFFF`, which should not take that much time, maybe a couple of hours.

majuss, 2022-06-16:
I found it, you mixed the TX and RX bin in your example. TX of the ESP has to go to the RX of the rf module.

enwi, 2022-06-17:
> I found it, you mixed the TX and RX bin in your example. TX of the ESP has to go to the RX of the rf module.

TX and RX are not mixed in the example. It is commonly known that RX needs to be connected to TX and vice versa, but since the code is written for the ESP and not the RF module the TX/RX pin definitions refer to the ESP. But to make this more clear I could add a comment above these definitions 🙂

majuss, 2022-06-17:
[image]

It is defintely mixed up in your comment. PIN 17 of the ESP32 is TX. And you say, in the comment behind, that this Pin is going to the TX of the RF module!

enwi, 2022-06-17:
Does it work now though?

enwi, 2022-06-17:
> It is defintely mixed up in your comment

Oh you are right, I didn't know that it said RX/TX pin of the RF module, I will correct the comment for any future users.

> The chinese support will send me a "universal ID"

So you have been in touch with the manufacturer? Interesting that there is sth. like a universal ID. Would be interesting if the inverter replies to that

majuss, 2022-06-26:
Soooo I have big news. I tried to brute force ID's starting from 45002800 but I had no luck. The chinese support got me the "universal ids" which didn't worked at all: 26002777 and 26002779. So this got me thinking and I started brute forcing from 26000000... And you wouldn't believe it. I found valid ids for my inverters: 45002818 has the ID 26002757 and 45002819 has 26002762.
I know compiled ESPhome and a custom sensor with your library which works flawless. In addition I replaced the RF module with the ESP32. Everything works and I know have the values directly in Home Assistant :)

closed 2022-06-26 by majuss

enwi, 2022-06-26:
Wow that's awesome 😊

But still weird that the IDs of your inverter don't match the inverter type afaik. But great that you could brute force the IDs. How long did it take to brute force btw?

majuss, 2022-06-26:
Not that long. I think I tried 1 h with 45* and 1-2 h with 26*.

WeareGermans, 2022-06-27:
Can you control the inverter, too?
As in changing the output volume and switching the power on and off?

majuss, 2022-06-27:
Sure thing it should work, why not?
But why would you want to change these parameters? It doesn't really make sense. Are you afraid because in germany there is only 600 W "allowed"?

WeareGermans, 2022-06-27:
Good question!
1. The design is flawed to not use the s.. word. My 700W inverter has 12.5A fuses soldered in line to each input. So 350W/24V= 14,6A. So 700W is not possible even in the optimal way of 2 strings of 350W in parallel.
2. I want to use it with a battery system and only switch it on at times where i am consuming more than it can deliver. (automated of cause) If i do not limit the power the fuse will (and already did ;-) ).

If you are already able to control it without the databox and software with only your ESP, that would help me very much! :-)

enwi, 2022-07-13:
@WeareGermans I for my part can tell you that my 2 inverters (both SG200MS) send back their status when using the ID's printed on them