atc1441/NETSGPClient #16: Daisy chaining UART of inverters

enwi, 2022-07-19:
Hey, your work is so great. I cant wait to test your project in the future. 
Is it posible to get rid of the RF modul an daisy chain the UART between the inverters?

_Originally posted by @detobsen in https://github.com/atc1441/NETSGPClient/issues/2#issuecomment-1188312946_

enwi, 2022-07-19:
I would say it is not possible, because UART is not designed to be daisy chaned unless you create a loop. For example the TX of one inverter goes to the RX of the next, this continues until the las one, where TX goes to the RX of your ESP8266/ESP32, where in turn the TX of the ESP goes to the RX of the first inverter. Since the inverters do not support retransmitting packages for other inverters this will not work and you are better of either putting an ESP directly on each inverter like in #14 or to use an LC12S to communicate wirelessly. Another thing to keep in mind is the operational safety of your inverters, adding an ESP internally is one thing, but adding cables to connect to other inverters also might impact the waterproofing.

enwi, 2022-07-19:
> Maybe a daisy chain is the wrong technic. So what about an asynch multidrop, we have only one master here.

Async Multi-Drop would be an option. Since there are different configurations I would go with this one:

[image]
> http://www.sampson-jeff.com/article/interface/int.htm

> On the other hand I don't want to put them outdoors, I will put them in a cabinet and use them as a stack of modular battery inverters. So I can scale the output power while my pv system is growing.

Just a side note, if you get the SGXXXMX (with medium operating voltage) make sure your system voltage (battery voltage) is at least 24V. Unless you get the SGXXXLX (with low operating voltage), which I haven't seen anywhere other than the manufacturere site, the inverter won't start operating.

detobsen, 2022-07-19:
Thanks for your note. I'm planing to have a 48V battery system and use the SG1400HQ. They have a input range between 36-60V and cost around 220€. By NETek directly the price seams 147€ ([https://de.enfsolar.com/pv/inverter-datasheet/12558]). So you have around 100€/kW.

enwi, 2022-07-19:
Galvanic isolation could be good, but afaik the inverters are only powered by the solar input and are somewhat isolated from the AC output

closed 2022-08-03 by enwi