atc1441/NETSGPClient #18: problem with arduino 1.8.19

goliath51, 2022-08-07:
How do change your source with arduino 1.8 because I have this probleme:
AsyncNETSGPClient.h:48:5: error: 'set' in namespace 'std' does not name a type
     std::set<uint32_t> mDevices; /// All devices to poll

enwi, 2022-08-07:
What architecture are you trying to compile for? `ESP8266`, `ESP32`, `AtMega328p`, ...

goliath51, 2022-08-07:
chipKit max32

enwi, 2022-08-07:
Does `max32` not support the standard c++ library?

enwi, 2022-08-07:
Try installing [ArduinoSTL](https://github.com/mike-matera/ArduinoSTL) and include it before `NETSGPClient`

goliath51, 2022-08-07:
it doesn't work because this is not avr architecture

goliath51, 2022-08-07:
is it possible to use esp01s for this example (I think no because there is no two serial port, i know) or I can use serial -> wifi?

enwi, 2022-08-07:
You could use an ESP01 with wireless serial (there are a few options, maybe take some inspiration from #13)

enwi, 2022-08-07:
Another option is to implement the std::set container yourself, but I would think that `ArduinoSTL` should work, because the c++ standard library should be architecture independent

goliath51, 2022-08-07:
error message:
ATTENTION : la bibliothèque ArduinoSTL prétend être exécutable sur la (ou les) architecture(s) avr, megaavr et peut être incompatible avec votre carte actuelle qui s'exécute sur pic32.

enwi, 2022-08-07:
Isn't that just a warning?
```
WARNING: the ArduinoSTL library claims to be executable on the avr, megaavr architecture(s) and may be incompatible with your current board which runs on pic32.
```
In my opinion it would be odd if it does not work, but I also never used a chipKit max32 before

goliath51, 2022-08-07:
I will try MQTT server with esp8266 #13 or directly solder on inverter #14

enwi, 2022-08-08:
@goliath51 If you only have a single inverter than using the code from #13 and soldering the ESP directly to the inverter (#14) would be the best option

enwi, 2022-11-14:
@goliath51 Any updates?

closed 2023-01-02 by enwi