atc1441/NETSGPClient #19: Problem Compiling Node MCU

damianburrin, 2022-08-14:
HI I can't get this to compile on a nodeMCU .  Getting an error about serial2 beiing used out of scope..

When i look at the ino file it does seem to try an call the object with serial2 but not declare this until the setup function.  Am i misunderstanding

Thanks
Damian

[image]

enwi, 2022-08-15:
@damianburrin The NodeMCU uses an ESP8266, which only has two Serial interfaces `Serial` and `Serial1`. In the example you will need to replace `Serial2` with `Serial`

enwi, 2022-08-15:
I will update the example to make sure noone has this issue again

damianburrin, 2022-08-15:
Brill

Thank you

commit 5fc2908b (2022-08-15): Use macros to use different Serials on different architectures

- closes #19
closed 2022-08-15 by enwi