atc1441/NETSGPClient #21: Pin Out Diagram

damianburrin, 2022-08-20:
Hi Aaron,

Is there any chance you could provide a pinout wiring diagram please.  I'm struggling to see which pins on the mcu connect to which on the rf module

It looks like 
3v3 to vcc
gnd to gnd
d8 to  rx
d7 to  tx
d6 to set

The code references Pin 16 and 17 for rx and tx and 4 got enable 

Just getting myself a little confused.

Any help appreciated

Thank you
Damian

enwi, 2022-08-20:
The pin definitions are only for the ESP32. If you take a look at the examples on an ESP8266 D1 mini RX goes to TX and TX to RX and D2 to set:
https://github.com/atc1441/NETSGPClient/blob/5fc2908b7b5840f771b4629758138eb2d7d4f373/examples/AsyncDemo/AsyncDemo.ino#L13-L14

damianburrin, 2022-08-20:
> The pin definitions are only for the ESP32. If you take a look at the examples on an ESP8266 D1 mini RX goes to TX and TX to RX and D2 to set:
> 
> https://github.com/atc1441/NETSGPClient/blob/5fc2908b7b5840f771b4629758138eb2d7d4f373/examples/AsyncDemo/AsyncDemo.ino#L13-L14

yeah i have esp32 and a nodemcu esp8266,  I'm trying the node but getting garbage output.  I'm clearly doing something wrong

[image]

enwi, 2022-08-20:
On the ESP8266 you need to connect a USB to Serial Adapter to pin D4, because the default Serial is used for the communication with the LC12S module, that is why you are seeing garbage.

damianburrin, 2022-08-20:
> On the ESP8266 you need to connect a USB to Serial Adapter to pin D4, because the default Serial is used for the communication with the LC12S module, that is why you are seeing garbage.

ok - i'll try to get it working with the ESP32 board i've got.  Pin positions are different but should be able to find a diagram to match up pin 16 and 17

enwi, 2022-08-20:
Yeah using an ESP32 sure is easier for testing things out, the problem here is that you theoretically could output the log messages on the default Serial interface as well, but in reality the inverters don't like to receive more or less random data and will go into fault state.

damianburrin, 2022-08-20:
Well getting closer

[image]

damianburrin, 2022-08-20:
Using Pin 4 to set pin but get not found error

[image]

damianburrin, 2022-08-20:
It's really hard to tell from the video but Is the CS and gnd pin here joined so they both go to ground?

[image]

enwi, 2022-08-20:
Yes chip select needs to be grounded, it is active low.

damianburrin, 2022-08-20:
> 

Cheers - i thought so, but still getting the same issue

enwi, 2022-08-20:
Did you try swapping RX/TX?

damianburrin, 2022-08-20:
Yeah – I’m just going to solder the connectors as they were just pinned in to make sure the connection is ok

damianburrin, 2022-08-20:
Hmm no better

This is my board
[image]

and this is how i've set my id

[image]

Any thoughts would be great

enwi, 2022-08-20:
I would check if nothing else is connected to the used pins and maybe use different ones if that is possible.

damianburrin, 2022-08-22:
Thank you - you guys are the best.
Fully working and data being displayed on the built in screen of my board (Hiltec Wifi 32)

[image]

closed 2022-08-22 by damianburrin