atc1441/NETSGPClient #23: Problem with SPIFFS_Test

GsDublin, 2022-10-20:
SPIFFS_Test stuck when I declare
```
NETSGPClient client(Serial2, PIN_LC12S_SET); /// NETSGPClient instance
```
at 
```
if (!SPIFFS.begin(FORMAT_SPIFFS_IF_FAILED)) {
```

function `esp_vfs_spiffs_register` will never finish.

This happens with ESP Core > 1.0.6

GsDublin, 2022-10-24:
My workaround is to call the constructor within setup()

Declaration:
```AsyncNETSGPClient *netsgp_client;```

setup():
```netsgp_client = new AsyncNETSGPClient(Serial2, PIN_LC12S_SET); // Defaults to fetch status every 2 seconds```
```netsgp_client->setStatusCallback(onInverterStatus);```

enwi, 2022-11-14:
I can't see how this bug relates to this library, since we are not messing with `SPIFFS`

enwi, 2022-11-14:
Also I cannot reproduce it on ESP Core > 1.06

closed 2023-01-02 by enwi