atc1441/NETSGPClient #25: Reading correct device temperature (SG1400MQ)

lienbacher, 2023-02-14:
Hello!
I'm running an SG1400MQ and have been using this library successfully and would like to improve some aspects. 
It seems the temperature that is captured as a unsigned int is not the actual temperature, or at least when it crosses 0° it jumps to a different value. Has anybody been able to reverse engineer these 8bits of temperature data?

enwi, 2023-02-18:
Thanks for the feedback on the `SG1400MQ` so I can update the list. @atc1441 did all the reversing, but did not have enough data I think for the temperature part. If you figure it out I am happy to adjust the code

enwi, 2023-02-18:
By the way can you also tell me in which range the ID of your inverter starts at?

Member0815, 2023-02-19:
I'm using the NETSGPClient library for my SG600MQ and it works well. Thanks to the author(s)
1) When sun rises or sets the communication to the SG600MQ sometimes doesn't work correctly. I added 3 modifications:
a) If header of the answer could not be found, I clear the whole input buffer to not trigger a set of following errors
b) I added some plausibility checks if current, voltage and power is in an reasonable range for my microinverter and 400W panels
c) I added a field "eff" for efficiency which is calculated by acPower divided by dcPower if != 0
2) I do not find reasonable interpretation for the fields "state" and "temperature". Following is a database export of my readings for the last weeks, maybe someone has more information about this fields.
[export.csv](https://github.com/atc1441/NETSGPClient/files/10777615/export.csv)

enwi, 2023-02-19:
> I'm using the NETSGPClient library for my SG600MQ and it works well. Thanks to the author(s)

Thanks 😊

> b) I added some plausibility checks if current, voltage and power is in an reasonable range for my microinverter and 400W panels

I actually did that as well, but I am not sure if we can add this to the library itself. For that we would need to be able to autodetect the inverter, which might be possible with the ID, but since the list is incomplete I would not do that

Member0815, 2023-02-20:
Hi Moritz,

 

That’s not a ChangeRequest, I only described, what I did ;-)

Autodection doesn’t make sense in my eyes. If really adding some code to the library then maybe the check itself based on some #defines for the range of power, voltage, current, so everybody can adapt this easily ;-)

 

The other point was “receiving nonsense bytes” while inverter has not enough power from the panels. I added the following to my code to clear the input buffer, because stream.find() seems not to do this – despite documentation:

 

bool NETSGPClient::findAndReadReply(const Command command) {

    // Search for a reply header consisting of magic byte and one of the command bytes

    const char header[2] = {MAGIC_BYTE, command};

    if (!mStream.find(header, 2)) {  //BH liest angeblich bis header gefunden oder Timeout 1000ms, scheint nicht zu stimmen

        while (mStream.available()) mStream.readBytes(&mBuffer[2], 1);   // BH Lese evtl. vorhandene Zeichen im Stream und ignoriere sie

        myLog.println(F("SGP: findAndReadReply: Could not find header"));

         return false;

    }

 

 

Some other aspects I added are:

-          Global variable for night or day in the main program based on sunrise / sunset

-          Global variable for MQTT-connection in Async-mode

Thus sending no status request to the inverter when no MQTT-connection or nightmode

 

Reducing the RF power setting of LC12S from 12 dBm to 0 dBm to be conform with German regulations …

 

Thanks again

lienbacher, 2023-04-28:
> By the way can you also tell me in which range the ID of your inverter starts at?

Hey @enwi, sorry for not coming back to this earlier. After all my Inverter died after 2 months, the dealer was not willing to fix the issue so I returned the device to amazon which thankfully worked like a treat. So I don't have this inverter anymore, but I just came across my old config and my ID started with 0x55000000.

enwi, 2023-05-01:
> Hey @enwi, sorry for not coming back to this earlier. After all my Inverter died after 2 months, the dealer was not willing to fix the issue so I returned the device to amazon which thankfully [...]

Sad to hear it broke that quickly, but thanks for the info 🙂

lienbacher, 2023-05-01:
even worse, there was zero customer support from the dealer, plus I got to say the hoymiles inverters are not only significantly more efficient, but also provide significantly more accurate and complete sensor data. Just in case anybody considering these inverters will read this. 
After all I'm honestly happy it broke in time!